stm32-rs/stm32-usbd #9: stm32-usbd shows surprising errors just by being mentioned in the Cargo.toml

therealprof, 2019-08-25:
stm32-usbd throws tons of errors just by being mentioned in the Cargo.toml without target feature. This makes it somewhat hard to use in a crate which targets multiple targets where USB support may actually not available. It would be nice to have a "no-op" target which still allows the crate the be in the dependencies even without being used. 

Further it would be a lot more user friendly if those non-saying error messages would actually hint at the real problem...

```
   Compiling stm32-usbd v0.3.1
error[E0432]: unresolved import `crate::target::UsbAccessType`
 --> /Users/egger/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-usbd-0.3.1/src/endpoint.rs:5:40
  |
5 | use crate::target::{UsbRegisters, usb, UsbAccessType};
  |                                        ^^^^^^^^^^^^^ no `UsbAccessType` in `target`

error[E0432]: unresolved imports `crate::target::UsbAccessType`, `crate::target::EP_MEM_SIZE`
 --> /Users/egger/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-usbd-0.3.1/src/endpoint_memory.rs:3:21
  |
3 | use crate::target::{UsbAccessType, EP_MEM_ADDR, EP_MEM_SIZE, NUM_ENDPOINTS};
  |                     ^^^^^^^^^^^^^               ^^^^^^^^^^^ no `EP_MEM_SIZE` in `target`
  |                     |
  |                     no `UsbAccessType` in `target`

error[E0432]: unresolved import `crate::target::USB`
 --> /Users/egger/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-usbd-0.3.1/src/bus.rs:9:21
  |
9 | use crate::target::{USB, apb_usb_enable, NUM_ENDPOINTS, UsbRegisters, UsbPins};
  |                     ^^^ no `USB` in `target`

error[E0432]: unresolved import `crate::target::usb_pins`
  --> /Users/egger/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-usbd-0.3.1/src/lib.rs:17:24
   |
17 | pub use crate::target::usb_pins::UsbPinsType;
   |                        ^^^^^^^^ could not find `usb_pins` in `target`

error[E0433]: failed to resolve: use of undeclared type or module `hal`
  --> /Users/egger/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-usbd-0.3.1/src/target.rs:58:31
   |
58 |         let rcc = unsafe { (&*hal::stm32::RCC::ptr()) };
   |                               ^^^ use of undeclared type or module `hal`

error[E0433]: failed to resolve: use of undeclared type or module `USB`
  --> /Users/egger/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-usbd-0.3.1/src/target.rs:76:19
   |
76 |         let ptr = USB::ptr() as *const Self::Target;
   |                   ^^^ use of undeclared type or module `USB`

error[E0433]: failed to resolve: use of undeclared type or module `USB`
  --> /Users/egger/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-usbd-0.3.1/src/target.rs:87:23
   |
87 |         let usb_ptr = USB::ptr() as *const usb::RegisterBlock;
   |                       ^^^ use of undeclared type or module `USB`

error[E0412]: cannot find type `USB` in this scope
  --> /Users/egger/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-usbd-0.3.1/src/target.rs:70:25
   |
70 | pub struct UsbRegisters(USB);
   |                         ^^^ not found in this scope

error[E0412]: cannot find type `USB` in this scope
  --> /Users/egger/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32-usbd-0.3.1/src/target.rs:82:21
   |
82 |     pub fn new(usb: USB) -> Self {
   |                     ^^^ not found in this scope

error: aborting due to 9 previous errors
```

Disasm, 2019-08-25:
You can make `stm32-usbd` optional and activate it only for those targets that support USB. But yes, I agree that these error messages are not helpful.

therealprof, 2019-08-25:
How do I specify dependencies based on features?

Disasm, 2019-08-25:
For example,
```toml
[dependencies]
stm32-usbd = { version = "0.3", optional = true }

[features]
stm32f042 = ["stm32f0xx-hal/stm32f042", "stm32-usbd", "stm32-usbd/stm32f042xx"]
stm32f099 = ["stm32f0xx-hal/stm32f099"]
```

therealprof, 2019-08-26:
@Disasm Thanks, that's certainly not properly explained in the `cargo` documentation (or I'm to daft to find it).

therealprof, 2019-08-27:
I think we can close this then.

closed 2019-08-27 by therealprof