stm32-rs/stm32-usbd #17: STM32F3Discovery fails to enumerate using 0.4 or 0.5 (works fine with 0.3)

Dr-Derivative, 2019-11-17:
This can be reproduced using the [stm32-usbd-examples](https://github.com/stm32-rs/stm32-usbd-examples) repo. If you follow the steps in the README.md file everything runs as expected since it uses stm32-usbd 0.3. 

If you edit Cargo.toml to upgrade stm32-usbd to 0.4 (and upgrade stm32f3xx-hal to 0.3 as required) the code compiles without any errors (although there are a few warnings about unused `Result`s). The device fails to enumerate correctly. This results in the "USB device not recognised" popup on Windows 10, and according to Device Manager "A request for the USB device descriptor failed". The same thing happens with 0.5.

I don't have much experience with USB stuff so I don't even know where to get started tracking this down, but I'm happy to help with a fix/PR.

Disasm, 2019-11-17:
I also faced with a problem like yours. Try adding this line: https://github.com/stm32-rs/stm32f3xx-hal/blob/bb974b428d7f3b46d284aa77cbff5af6f90c31b2/examples/usb_serial.rs#L24 and removing `configure_usb_clock();` line

Dr-Derivative, 2019-11-17:
Indeed, that does work. Thank you!

Seems like the examples repo should be updated then.

Unfortunately, my discovery board is the only stm32 board I have to test on, so I can't say if the other examples work with the latest stm32-usbd version. But I can put together a PR for just the f303 example.

Disasm, 2019-11-17:
While moving to 0.5 I noticed this problem with f303 only, other boards worked well. I will update the examples repo.

Dr-Derivative, 2019-11-17:
Alright, thanks!

I'll close this issue then.

closed 2019-11-17 by Dr-Derivative

Disasm, 2019-11-17:
Fixed and tested: https://github.com/stm32-rs/stm32-usbd-examples/commit/3a37939c428ffac9e16b9a0c06111df36f254bac