stm32-rs/stm32-usbd #19: Support for STM OTG_FS core present in STM32F4xx (and higher) series microcontrollers

thousandeyes-fwieland, 2019-12-18:
Eg. the STM32F446xx family has OTG_FS and OTG_HS cores which can function both as peripheral and host (for USB OTG). I would like to be able to use this crate with the OTG_FS core in peripheral mode. I can help with the implementation, but will need some pointers on where to start.

I'm currently trying to find out more about the OTG_FS core itself.

TeXitoi, 2019-12-18:
There is a WiP here: https://github.com/Disasm/stm32f429-usbd-fs

Disasm, 2019-12-18:
I think that driver for OTG_FS&HS should live in a different crate for multiple reasons:
* These peripherals are completely different and require different device-specific parameters
* OTG_FS peripheral is present not only in STM32, but also in GD32VF103. I'm going to rename `stm32f429-usbd-fs` into something like `synopsys-usb`

@thousandeyes-fwieland Feel free to join `stm32f429-usbd-fs` development. At the moment, that driver works for STM32F429 and STM32F401 MCUs, but there are a lot of things to do:
* Separating the driver code from the hal libraries in the same way as it done in `stm32-usbd`.
* Replacing `hs`/`fs` features with configuration flag. Ideally, user should be able to use this driver for both HS and FS peripherals at the same time (F429 has both).
* Fix `stm32-rs` patch set to support more devices.
* Investigate the differences between F429 and the other F4xx with the updated USB OTG IP (for example, F413). Add support for these chips.

TeXitoi, 2019-12-19:
@Disasm you make it work on GD32VF103?

Disasm, 2019-12-19:
@TeXitoi I tried to plug this library in for GD32VF103, but it didn't work out of the box. I'm planning to add GD32VF103 support anyway.

Disasm, 2020-10-01:
Closing the issue as `synopsys-usb-otg` was released and a lot of problems solved since then.

closed 2020-10-01 by Disasm