stm32-rs/stm32-usbd #26: MIT license compliance w.r.t. embedded applications

agausmann, 2021-01-29:
:"Original" issue is at mvirkkunen/usb-device#52, but I wanted to open an issue here as well, because this crate is also MIT-licensed.

I would love to use this in my embedded projects, but I'm not comfortable using MIT-licensed code in such applications, because of the condition:

> The above copyright notice and this permission notice shall be included in all
copies or substantial portions of the Software.

Granted, it does just say "included", not "displayed", so I guess I could include the license somewhere in the firmware binary and it would be okay? But I feel like being able to view the license is somewhat implied, so I'm not satisfied with this.

In many embedded applications, there's no easy way to prominently "display" license notices like this, and I think the (implied) obligation to do so is an unnecessary burden. For example, if I'm writing firmware for a simple keyboard or mouse, how would I display the license notice from the device?

So, I think MIT is not a great fit for libraries used in the embedded ecosystem. Alternatively, I would suggest the Apache-2.0 license, as that is the one most used by the Rust Embedded WG, and only requires a license notice for copies of the source code form. Also, to not "break compatibility" with the current licensees, I would suggest retaining the MIT license, and dual-licensing with Apache-2.0.

Disasm, 2021-01-29:
@agausmann Why do you think the copyright notice text should be included in the *firmware binary*? Why not put the license into a separate file and distribute it along with the firmware binary?

agausmann, 2021-01-30:
You could argue that a firmware binary flashed to a device is a "copy of the Software". If the license notice is a separate file, it is not included when the firmware is flashed.

camrbuss, 2021-02-21:
> It could be argued that a firmware binary flashed to a device is a "copy of the Software". If the license notice is a separate file, it is not included when the firmware is flashed.

Could you include a printed copy (or QR code link) of the MIT License in the product you are shipping? Would this suffice? (Note: I am not a contributor to this repo, just in a similar situation)

therealprof, 2021-02-22:
> Could you include a printed copy (or QR code link) of the MIT License in the product you are shipping? Would this suffice? (Note: I am not a contributor to this repo, just in a similar situation)

Yes, that's how it usually works for retail hardware using open source as well.