stm32-rs/stm32-usbd #35: Request release

David-OConnor, 2024-02-24:
So deps can release on Crates.io. Thx!

tomioe, 2024-03-14:
+1 

The examples linked in this repo's README to the HAL Examples are no longer compiling, since the crates.io version (last release 0.6.0) has an old dependency for `usb-device`.

dlaw, 2024-03-29:
I am successfully using this version of stm32-usbd on real hardware (STM32G473CCT6) with the following Cargo.toml override:
```
[patch.crates-io]
stm32-usbd = { git = "https://github.com/stm32-rs/stm32-usbd.git" }
```
as well as the change to use StringDescriptors as described by tomioe.

funbiscuit, 2024-04-24:
I confirm current version from master is working properly on blue pill (STM32F103C8T6) with changes from comments above. Tested as virtual com port. Should we expect new version released any time soon?

David-OConnor, 2024-05-08:
What's the word? Should I fork and release this as a new crate?

David-OConnor, 2024-05-08:
I published a fork; same as current master, but on crates.io. This allows you to use it as a dependency of a different released crate. https://crates.io/crates/stm32-usbd2

```toml
stm32-usbd = { package = "stm32-usbd2", version = "^0.7.0" }
```

sunsided, 2024-06-29:
Just stumbled over the same issue, @dlaw's solution [here](https://github.com/stm32-rs/stm32-usbd/issues/35#issuecomment-2027747185) also helped me for the time being.

sunsided, 2024-06-29:
> I published a fork; same as current master, but on crates.io. This allows you to use it as a dependency of a different released crate. https://crates.io/crates/stm32-usbd2
> 
> ```toml
> stm32-usbd = { package = "stm32-usbd2", version = "^0.7.0" }
> ```

@David-OConnor Make it a pull request! 😬

David-OConnor, 2024-06-29:
What do you mean?

sunsided, 2024-06-30:
@David-OConnor Might be easier to get a release here if all it takes is merging the version bump. I'm just hoping.

David-OConnor, 2024-06-30:
It doesn't take any merge - whoever owns the crate just needs to do it.